platformio/platform-renesas-ra #1: Uno R4 Minima upload doesn't work

maxgerhardt, 2023-07-07:
```
CURRENT: upload_protocol = dfu
CURRENT: upload_protocol = dfu
"C:\Users\Max\.platformio\packages\tool-dfuutil\bin\dfu-util" -d 0x2341:0x0069,0x2341:0x0369 -a 0 -s 0x4000:leave -D "C:\Users\Max\Desktop\ren\example\.pio\build\uno_r4_minima\firmware.bin"
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 2341:0369
Device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Warning: DfuSe option used on non-DfuSe device
Device returned transfer size 64
DfuSe interface name: "CodeFlash "
Downloading element to address = 0x00004000, size = 70240

Erase           [                         ]   0%            0 bytes
Erase           [=                        ]   4%         2816 bytes
Erase           [==                       ]   8%         5632 bytes
Erase           [===                      ]  12%         8448 bytes
Erase           [====                     ]  16%        11264 bytes
Erase           [=====                    ]  20%        14080 bytes
Erase           [======                   ]  24%        16896 bytes
Erase           [=======                  ]  28%        19712 bytes
Erase           [========                 ]  32%        22528 bytes
Erase           [=========                ]  36%        25344 bytes
Erase           [==========               ]  40%        28096 bytes
Erase           [===========              ]  44%        30912 bytes
Erase           [============             ]  48%        33728 bytes
Erase           [=============            ]  52%        36544 bytes
Erase           [==============           ]  56%        39360 bytes
Erase           [===============          ]  60%        42176 bytes
Erase           [================         ]  64%        44992 bytes
Erase           [=================        ]  68%        47808 bytes
Erase           [==================       ]  72%        50624 bytes
Erase           [===================      ]  76%        53440 bytes
Erase           [====================     ]  80%        56192 bytes
Erase           [=====================    ]  84%        59008 bytes
Erase           [======================   ]  88%        61824 bytes
Erase           [=======================  ]  92%        64640 bytes
Erase           [======================== ]  96%        67456 bytes
Erase           [=========================] 100%        70240 bytes
Erase    done.

Download        [                         ]   0%            0 bytes

```
Then it hangs forever. 

Likely this is becuase
1. The wrong dfu-util version is used, it really really needs that one referenced in https://downloads.arduino.cc/packages/package_index.json (`"version": "0.11.0-arduino5", "name": "dfu-util"`). PlatformIO built in one has problems on some systems (https://github.com/maxgerhardt/platform-renesas/issues/2)
2. `-Q` flag is not given for final reset

maxgerhardt, 2023-07-07:
It also stands that the build script should **not** add a DFU suffix to the output, becasue the Arduino IDE doesn't do that too.

maxgerhardt, 2023-07-07:
For refernece: Verbose Upload log from my platform:
```
CURRENT: upload_protocol = dfu
"C:\Users\Max\.platformio\packages\tool-dfuutil\dfu-util" -d 0x2341:0x0069,0x2341:0x0369 -a 0 -Q -D "C:\Users\Max\Desktop\ren\example\.pio\build\uno_r4_minima\firmware.bin"
dfu-util 0.11-arduino4

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 2341:0069
Run-Time device DFU version 0101
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Device returned transfer size 64
Copying data from PC to DFU device

Download        [                         ]   0%            0 bytes
Download        [=                        ]   4%         1408 bytes
Download        [==                       ]   8%         2752 bytes
Download        [===                      ]  12%         4096 bytes
Download        [====                     ]  16%         5440 bytes
Download        [====                     ]  19%         6528 bytes
Download        [=====                    ]  20%         6784 bytes
Download        [======                   ]  24%         8128 bytes
Download        [=======                  ]  28%         9472 bytes
Download        [========                 ]  32%        10816 bytes
Download        [=========                ]  36%        12160 bytes
Download        [=========                ]  39%        13312 bytes
Download        [==========               ]  40%        13504 bytes
Download        [===========              ]  44%        14848 bytes
Download        [============             ]  48%        16192 bytes
Download        [=============            ]  52%        17536 bytes
Download        [==============           ]  56%        18880 bytes
Download        [==============           ]  59%        20032 bytes
Download        [===============          ]  60%        20224 bytes
Download        [================         ]  64%        21568 bytes
Download        [=================        ]  68%        22912 bytes
Download        [==================       ]  72%        24256 bytes
Download        [===================      ]  76%        25600 bytes
Download        [===================      ]  79%        26688 bytes
Download        [====================     ]  80%        26944 bytes
Download        [=====================    ]  84%        28288 bytes
Download        [======================   ]  88%        29632 bytes
Download        [=======================  ]  92%        30976 bytes
Download        [======================== ]  96%        32320 bytes
Download        [======================== ]  99%        33408 bytes
Download        [=========================] 100%        33584 bytes
Download done.
DFU state(7) = dfuMANIFEST, status(0) = No error condition is present
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
<lambda>(["upload"], [".pio\build\uno_r4_minima\firmware.bin"])
============== [SUCCESS] Took 15.56 seconds ==============
```

valeros, 2023-07-07:
Hi @maxgerhardt, many thanks for your feedback. I've pushed some fixes to the dev branch, it would be great if you could test out the updated version.

maxgerhardt, 2023-07-07:
Works.
```
CURRENT: upload_protocol = dfu
"C:\Users\Max\.platformio\packages\tool-dfuutil-arduino\dfu-util" -d 0x2341:0x0069,0x2341:0x0369 -a 0 -Q -D "C:\Users\Max\Desktop\ren\example\.pio\build\uno_r4_minima\firmware.bin"
dfu-util 0.11-arduino4
[...]
Download        [======================== ]  99%        69952 bytes
Download        [=========================] 100%        70240 bytes
Download done.
DFU state(7) = dfuMANIFEST, status(0) = No error condition is present
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
============== [SUCCESS] Took 23.17 seconds ==============
```

closed 2023-07-07 by maxgerhardt