platformio/platform-renesas-ra #2: Build flags missing /wrong from Arduino

maxgerhardt, 2023-07-07:
Example PlatformIO invocation:
>arm-none-eabi-g++ -o .pio\build\uno_r4_minima\FrameworkArduino\api\Print.cpp.o -c -Waggregate-return -Wall -Wconversion -Wextra -Wfloat-equal -Wlogical-op -Wmissing-declarations -Wpointer-arith -Wshadow -Wuninitialized -Wunused -fdata-sections -ffunction-sections -fmessage-length=0 -fsigned-char -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -iprefixC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno @C:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\variants\MINIMA\includes.txt -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DPLATFORMIO=60108 -DARDUINO_UNOR4_MINIMA -DLWIP_DNS=1 -DCFG_TUSB_MCU=OPT_MCU_RAXXX 
-D_RA_CORE=CM4 -D_RENESAS_RA_ -DARDUINO=10810 -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -DF_CPU=48000000L -IC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\cores\arduino\tinyusb -IC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\cores\arduino -IC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\cores\arduino\api\deprecated -IC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\cores\arduino\api\deprecated-avr-comp -IC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\variants\MINIMA C:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\cores\arduino\api\Print.cpp

Example Arduino IDE invocation:

>"C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -MMD -nostdlib -DF_CPU=48000000 -DARDUINO_UNOR4_MINIMA -MMD -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -DARDUINO=10819 "-DPROJECT_NAME=\"C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766/Blink.ino\"" -DARDUINO_MINIMA -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1/cores/arduino/tinyusb" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\cores\\arduino" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA" "-iprefixC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1" "@C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA/includes.txt" "C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\cores\\arduino\\api\\Print.cpp" -o "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\core\\api\\Print.cpp.o"

What PlatformIO misses:
1: `-Os` (makes firmware **really* big!)
2. `-g3`
3. `-fno-use-cxa-atexit`
4. `-fno-rtti`
5. `-fno-exception`
6. `-nostdlib`

What's wrong:
1. `-DLWIP_DNS=1` given for Minima. Only used with Uno R4 WiFi
2. Different warning flags?

In the Linking phase, PlatformIO:
>arm-none-eabi-g++ -o .pio\build\uno_r4_minima\firmware.elf -T C:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\variants\MINIMA\fsp.ld -mcpu=cortex-m4 -mthumb -Wl,--gc-sections --specs=nosys.specs -mfloat-abi=hard -mfpu=fpv4-sp-d16 .pio\build\uno_r4_minima\src\main.cpp.o -L.pio\build\uno_r4_minima -LC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\variants\MINIMA -LC:\Users\Max\.platformio\packages\framework-arduinorenesas-uno\variants\MINIMA\libs -Wl,--start-group .pio\build\uno_r4_minima\libFrameworkArduinoVariant.a .pio\build\uno_r4_minima\libFrameworkArduino.a -lfsp -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group

Arduino IDE:
>"C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -Wl,--gc-sections --specs=nosys.specs -w -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -o "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766/Blink.ino.elf" "-LC:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766" "-LC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA" "-TC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA/fsp.ld" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\sketch\\Blink.ino.cpp.o" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\core\\tmp_gen_c_files\\common_data.c.o" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\core\\tmp_gen_c_files\\main.c.o" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\core\\tmp_gen_c_files\\pin_data.c.o" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766\\core\\variant.cpp.o" -Wl,--start-group "C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.1\\variants\\MINIMA/libs/libfsp.a" "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766/core\\core.a" --specs=nano.specs -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group "-Wl,-Map,C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_350766/Blink.ino.map"

What's missing:
* no `.map` file generation

In general:
* Arduino IDE uses GCC 7.0.1, PlatformIO uses 10.3.1. May have subtle differences.

maxgerhardt, 2023-07-07:
Oh, and actually a really critical thing too: The `-std=c++17` is missing for CXXFLAGS and `-std=gnu11` is missing in `CLFAGS`. It doesn't set the C/C++ standard version.

valeros, 2023-07-07:
Thanks for a thorough review. 

> What PlatformIO misses:

I've added flags except `-g3` as using it in release mode is quite debatable and `-nostdlib` added to `LINKFLAGS`  as it's a linker option. 

> -DLWIP_DNS=1 given for Minima. Only used with Uno R4 WiFi

Removed

> Different warning flags?

Flags are probably the same, the difference is that we're importing them into the build environment, so they're visible in CLI.

> no .map file generation

Added.

> Arduino IDE uses GCC 7.0.1, PlatformIO uses 10.3.1. May have subtle differences.

I believe it shouldn't be an issue.

> The -std=c++17 is missing for CXXFLAGS and -std=gnu11 is missing in CLFAGS.

Added.

maxgerhardt, 2023-07-07:
Looking all good now!

closed 2023-07-07 by maxgerhardt