platformio/platform-renesas-ra #5: Intellisense issue on some systems

maxgerhardt, 2023-07-07:
Per https://github.com/maxgerhardt/platform-renesas/issues/1#issuecomment-1625310655.

Magically on this user's system, the `c_cpp_properties.json` gets wrongly generated with missing path seperator character:

[image]

I have no idea how that's possible since it does work on my Windows 10, x64, system:

```json
            "compilerPath": "C:/Users/Max/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc.exe",
            "compilerArgs": [
                "-mcpu=cortex-m4",
                "-mfloat-abi=hard",
                "-mfpu=fpv4-sp-d16",
                "-mthumb",
                "-iprefixC:/Users/Max/.platformio/packages/framework-arduinorenesas-uno",
                "@C:/Users/Max/.platformio/packages/framework-arduinorenesas-uno/variants/UNOWIFIR4/includes.txt",
                ""
            ]
```

I pushed a countermeasure for this, that is, in the `__idedata` phase, don't return `-iprefix <prefx> @<file>`, but actually expand out all include paths:  https://github.com/maxgerhardt/platform-renesas/blob/bc8d2f1ae4d0a8507dbee06861a884293c0d83a2/builder/frameworks/arduino/mbed-core/arduino-core-mbed.py#L157-L179

According to the user, the problem remained present even after reinstalling everything and trying it on a different computer. Might be issues with specific character encoding sets, e.g. Chinese? I have no idea real idea though.

closed 2023-07-08 by valeros

maxgerhardt, 2023-07-08:
@LYB030 can you again retry with [PlatformIO CLI](https://github.com/Community-PIO-CH32V/ch32-pio-projects#update-via-platformio-core-cli) -> 
```
pio pkg update -g -p renesas-ra
```
and Ctrl+Shift+P -> Rebuild Intellisense.

Is the Intellisense issue still there?

LYB030, 2023-07-09:
Bug solved, thank you for your help！

commit a727d324 (2023-07-08): Temprarily fix iprefix flag

Resolves #5