platformio/platform-renesas-ra #10: RTC.begin() deadlocks processor

maxgerhardt, 2023-08-06:
See https://github.com/arduino/ArduinoCore-renesas/issues/88.

The Arduino framework code has a bug / undefined behavior in which the `RTClass::begin()` function, when compiled by the here used ARM GCC 10.3.1. produces wrong assembly in such a way that it causes an infinite loop at the `is_initialized = false` statement. Calling `RTC.begin();` is a game-over.

Broke a user trying to use an Arduino IoT sketch with the Uno R4 WiFi at https://community.platformio.org/t/cant-connect-to-arduino-iot-cloud-with-platformio/35087. 

This either needs a minimal package update with just the patch or (sadly) a downgrade of the compiler version to exactly match the Arduino IDE's behavior.

maxgerhardt, 2023-08-06:
**ANOTHER** Bug of the same kind, a missing `return` of a boolean value, causes a Usage fault **crash** of the processor in the `WiFiUDP::read_if_needed()` function, as explained [in the topic](https://community.platformio.org/t/cant-connect-to-arduino-iot-cloud-with-platformio/35087/8?u=maxgerhardt).

This bug is however not present in the 1.0.2 version of the core (function has been renamed and is `void` now), but it's present in the 1.0.1 version that PlatformIO is using.

valeros, 2023-08-07:
Thanks for pointing it out. I've uploaded v1.0.2 with your patch fix included.

commit a0d14d9b (2023-08-07): Update Arduino core to v1.0.2

Resolves #10
closed 2023-08-07 by valeros