platformio/platform-renesas-ra #12: Upgrade framework-arduinorenesas-uno

runger1101001, 2023-09-19:
https://github.com/arduino/ArduinoCore-renesas/tree/1.0.4 has been recently released, but platform-renesas-ra 1.1.0 is using framework-arduinorenesas-uno in the 1.0.2 version.

Is there a way to override this from the platformio.ini?

Is this even the correct project to report this bug? Where I can I find the definition of framework-arduinorenesas-uno?

Thank you!

valeros, 2023-09-20:
Thanks for reporting, updated.

closed 2023-09-20 by valeros

runger1101001, 2023-09-20:
```
runger@Richards-Pro-2 minima_pwm_test % pio pkg update -t framework-arduinorenesas-uno                             
Resolving uno_r4_minima dependencies...
Tool Manager: Updating framework-arduinorenesas-uno @ 1.0.2
Tool Manager: Removing framework-arduinorenesas-uno @ 1.0.2
Tool Manager: framework-arduinorenesas-uno@1.0.2 has been removed!
Tool Manager: Installing platformio/framework-arduinorenesas-uno @ 1.0.4
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: framework-arduinorenesas-uno@1.0.4 has been installed!
```

❤️ Thanks so much for the super-fast response!

commit 494b0de2 (2023-09-20): Update Arduino packages to v1.0.4

Resolves #12