platformio/platform-renesas-ra #16: Usage fault when executing code complied with GCC 10 but not with GCC 7

djix123, 2023-10-11:
The following example causes a usage fault when built with GCC 10, but reverting back to GCC 7 fixes the problem. The fault occurs when executing the 2nd delay in ```loop()```, or directly thereafter.

<code>
#include  < Arduino.h >
#include "Arduino_LED_Matrix.h"

ArduinoLEDMatrix matrix;

void setup() {
    Serial.begin(115200);
    matrix.begin();
}

const uint32_t happy[] = {
        0x19819,
        0x80000001,
        0x81f8000
};
const uint32_t heart[] = {
        0x3184a444,
        0x44042081,
        0x100a0040
};

void loop(){
    matrix.loadFrame(happy);
    delay(500);
    matrix.loadFrame(heart);
    delay(500);
}
</code>

with ```platform.ini```

<code>
[env:uno_r4_wifi]
platform = renesas-ra
;platform = https://github.com/platformio/platform-renesas-ra.git@1.0.2
board = uno_r4_wifi
framework = arduino
</code>
<br>
and stack trace:
<br>
<code>
Firmware name: PROJECT_NAME, compiled on: Oct 10 2023
Fault on interrupt or bare metal(no OS) environment
===== Thread stack information =====
  addr: 20007ec0    data: 00000000
  addr: 20007ec4    data: 0000a500
  addr: 20007ec8    data: ffffffff
  addr: 20007ecc    data: 40046f00
  addr: 20007ed0    data: 00000000
  addr: 20007ed4    data: 0000440f
  addr: 20007ed8    data: 200009e8
  addr: 20007edc    data: 00006d89
  addr: 20007ee0    data: 0000d29c
  addr: 20007ee4    data: 00006dcf
  addr: 20007ee8    data: 0000d29c
  addr: 20007eec    data: 0000ce43
  addr: 20007ef0    data: 0000d29c
  addr: 20007ef4    data: 0000a8ff
  addr: 20007ef8    data: 0000a8f5
  addr: 20007efc    data: 00002599
====================================
=================== Registers information ====================
  R0 : 00000001  R1 : 0007a120  R2 : ffffffff  R3 : 0000416d
  R12: 00000003  LR : 00006e3f  PC : 200000b4  PSR: 20000000
==============================================================
Usage fault is caused by attempts to switch to an invalid state (e.g., ARM)
Show more call stack info by run: addr2line -e PROJECT_NAME.elf -a -f 200000b4 00006e3e 0000440e 00006d88 00006dce 0000ce42 0000a8fe 0000a8f4
</code>
and ```addr2line```:
<code>
0x200000b4
__bss_start__
.//SynologyDrive/Documents/MCU/projects/Arduino Uno R4 Wifi/src/main.cpp:18
0x00006e3e
delay
.//.platformio/packages/framework-arduinorenesas-uno/cores/arduino/time.cpp:9
0x0000440e
setup
.//SynologyDrive/Documents/MCU/projects/Arduino Uno R4 Wifi/src/main.cpp:22
0x00006d88
_Z12arduino_mainv
.//.platformio/packages/framework-arduinorenesas-uno/cores/arduino/main.cpp:117 (discriminator 1)
0x00006dce
atexit
.//.platformio/packages/framework-arduinorenesas-uno/cores/arduino/main.cpp:139
0x0000ce42
main
.//.platformio/packages/framework-arduinorenesas-uno/variants/UNOWIFIR4/tmp_gen_c_files/main.c:7
0x0000a8fe
Reset_Handler
/ssd/products-Eslov/FW/MainPrograms/hardware/arduino-git/renesas/extras/e2studioProjects/Santiago/Debug/../ra/fsp/src/bsp/cmsis/Device/RENESAS/Source/startup.c:67
0x0000a8f4
Reset_Handler
/ssd/products-Eslov/FW/MainPrograms/hardware/arduino-git/renesas/extras/e2studioProjects/Santiago/Debug/../ra/fsp/src/bsp/cmsis/Device/RENESAS/Source/startup.c:62
</code>

valeros, 2023-10-11:
Hi @djix123, could you please also test with the latest GCC v12?
```ini
[env:uno_r4_wifi]
platform = renesas-ra
;platform = https://github.com/platformio/platform-renesas-ra.git@1.0.2
board = uno_r4_wifi
framework = arduino
platform_packages =
    toolchain-gccarmnoneeabi @ ~1.120301.0
```

djix123, 2023-10-11:
I tried with GCC v12. I faults out even earlier - shortly after the call to hal_entry()

valeros, 2023-11-07:
Hi there, could you please also try the following toolchain package:
```ini
[env:uno_r4_wifi]
platform = renesas-ra
framework = arduino
board = uno_r4_wifi
platform_packages = 
    toolchain-gccarmnoneeabi @ 1.70201.0

```