platformio/platform-renesas-ra #17: Add support for WeAct Studio RA4M1 Board

djix123, 2023-10-25:
Please would you consider adding support for the new WeAct Studio RA4M1 board:

https://github.com/WeActStudio/WeActStudio.RA4M1_64Pin_CoreBoard

Thanks!

arwidcool, 2023-10-30:
I second this one. 

I ordered one. If I can set up the board correctly I will link it for someone to check over.

djix123, 2023-10-31:
I've had some success by using the UNO R4 Minima, and using JLink. Not perfect, but can get it going.

arwidcool, 2023-11-29:
> I've had some success by using the UNO R4 Minima as the board selection, and using JLink. Not perfect, but can get it going.

Having some issues initializing a blink sketch, Says programming is successful . The pin mapping on the LED corresponds to "pin 21" on the Arduino. Do you have your blink code?

djix123, 2023-12-01:
This is the code (very standard):
<code>
void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
}
//
void loop() {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
}
</code>

with ```platformio.ini```
<code>
[env:uno_r4_wifi]
platform = renesas-ra
board = uno_r4_wifi
framework = arduino
platform_packages = toolchain-gccarmnoneeabi@1.70201.0
</code>

arwidcool, 2023-12-02:
> That sounds good. In the end I started playing with the RASC directly:
> 
> https://github.com/djix123/WeActRA4M1CMake.git
> 
> https://github.com/djix123/WeActRA4M1RASC.git

Using the the Minima board and flashing with jlink when setting the cpu freq i was able to get the USB working no issues btw, so it has all the features and you can expose the pins.