platformio/platform-renesas-ra #18: Simple sketch using LED matrix doesn't work. void loop() is never called

Nefarious19, 2023-11-04:
Hello,

I tried to write some code on Arduino UNO R4 WiFi and it desn't work. I was debugging code and it seems that after enabling timer irq it is invoked all the time so program cannot go to loop function. But I'm not sure that is the case here.

code:

`
#include <Arduino.h>
#include <Arduino_LED_Matrix.h>

ArduinoLEDMatrix matrix;

uint32_t frame1 [3] =
{
  0,
  0,
  0
};

uint32_t frame2[3] =
{
  UINT32_MAX,
  UINT32_MAX,
  UINT32_MAX
};

void setup() {
  Serial.begin(115200);
  matrix.begin();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  matrix.loadFrame(frame2);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  matrix.loadFrame(frame1);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
`

the same code works fine in newest Arduino IDE 2.2.1

eyyyyyyy3, 2023-11-04:
I have the same problem. I got it working once but the loop only did one iteration before not working anymore.

valeros, 2023-11-06:
Hi there, could you please try downgrading the toolchain package to v7.2.1? For example:
```ini
[env:uno_r4_wifi]
platform = renesas-ra
framework = arduino
board = uno_r4_wifi
platform_packages = 
    toolchain-gccarmnoneeabi @ 1.70201.0

```

eyyyyyyy3, 2023-11-08:
It worked but is the solution really downgrading? Is the compiler broken or is there some linking problem with the compiler and PlatformIO and the Arduino framework?

valeros, 2023-11-09:
> It worked but is the solution really downgrading? Is the compiler broken or is there some linking problem with the compiler and PlatformIO and the Arduino framework?

The Arduino team decided to stick to GCC v7 for some reason, it seems we will need to downgrade the toolchain package by default as well.

eyyyyyyy3, 2023-11-10:
So this issue can be closed

commit ae32793a (2023-11-13): Downgrade GCC to v7

The toolchain version now corresponds to GCC used by the
original Arduino core to avoid compatibility issues

Resolves #18
closed 2023-11-13 by valeros